qkitgroup/qkit #2: VNA drivers

mpfirrmann, 2017-05-04:
Check VNA drivers for correct use of 
edel vs. port extension
estimating sweep time vs. trigger
set_Average

Schneider1, 2017-05-05:
To enable the new spectroscopy script, all VNAs need the following functions:

`pre_measurement` is called once at the start
`start_measurement` is called every time, a new set of averages has to be taken (e.g. reset averages)
`ready` is queried every 0.2s and returns a Boolean
`post_measurement` brings the VNA back to a mode where it is usable by its frontend (e.g. not in hold mode)

JBraumueller, 2017-08-03:
* set_Average is working now with Anritsu VNA MS4642B
* added parameter sweep_mode, replacing the hold function

These changes are comitted to the stable branch currently.

rotzinger, 2018-01-27:
What is the status here? Can we close this?

rotzinger, 2018-07-04:
There seems no hot interest anymore. Closing it.

closed 2018-07-04 by rotzinger