qkitgroup/qkit #6: qtlab dependance

mpfirrmann, 2017-05-04:
Figure out how much (and what) of qtlab we still do need

yschoen, 2017-05-12:
Dependency on pygtk was resolved by removing references to gobject, emit calls and the qtlab gui (will be replaced by gviewkit later).
Qtlab now starts from the core folder without pygtk and also on 64bit Python 2.7.
First tries creating instruments and controlling a microwave source as well as the spectrum analyzer were successful.
An overview of the changes made can be seen in core/README.

yschoen, 2017-05-12:
Which installation requirements are still exist? See attached old installation notes and include into qkit readme.
[readme_012015.txt](https://github.com/qkitgroup/qkit/files/996072/readme_012015.txt)

yschoen, 2017-05-12:
The core should now be tested during measurements. To switch from qtlab to the new core, only two changes are necessary:
- checkout the qtlab_core branch on the measurement PC (and pull latest version)
- change the start notebook from
  %run C:\<qtlab_root_folder>\source\qtlab_shell.py
  to:
  %run C:\qkit\qkit\core\source\qtlab_shell.py

mpfirrmann, 2017-05-26:
The Measurement-class needs certain functions from the qt.instrument-class:
get_parameter_names()
get_parameter_options()
get()
set()
and from the qt.instruments-class:
get()
get_instruments()

rotzinger, 2018-01-27:
This is (mostly) fixed. Please open another more specific issue, if something is not right.

closed 2018-01-27 by rotzinger