qkitgroup/qkit #26: Faulty X Axis Scaling in Qviewkit

fr34q, 2018-01-13:
When using logarithmical spaced values as x axis in a measurement (e.g. for a T1 measurement) the viewer interprets the x axis as lineary spaced thus leading to wrong axis labelling.

Example:
data0 > delay:
[image]
(x axis is not really the delay but index)
This leads to a T1 measurement looking like:
data0 > phase_avg_0
[image]
instead of an exponential decay

Expected behavior: When x axis is labelled "delay" it should also use the values given in the delay dataset and not extrapolate from only the first two data points or so.

rotzinger, 2018-01-15:
Hi Richard,

this is most likely not a bug. Datasets in data0/analysis0 are displayed in Qviewkit as bare dataset with the x-axis being either None (just the index) or a optional x0+dx*N pair.  If you want to display a dataset vs a different axis  use the add_view() function in qkit.storage.store , which creates a view on the data.

Cheers, Hannes

closed 2018-01-15 by rotzinger

fr34q, 2018-01-15:
Hi Hannes,

shouldn't we then at least update the x axis labelling accordingly to avoid confusion? In my case, I expected to see an exponential decay and it took me a while to realize what was happening here because the x axis label was given as "delay (ns)".

But as we typically set the x axis explicitly I would find it more intuitive to use if the hdf data container would then contain these views automatically. Alternatively, a nice solution would be if one could specify an axis dataset for each individual dataset which is then used instead of linear spacing.

Your thoughts?

See you, Richard

Schneider1, 2019-02-25:
Although considered to be not a bug, this behavior should have changed. Can you confirm that, Richard?

fr34q, 2019-02-25:
Yes, as I said last week, this now works as I would expect it.

Delays for T1 measurement:
[image]

Plot of the averaged phase:
[image]

So issue can stay closed ;)