qkitgroup/qkit #27: check if pivisa-py is an option for qkit

rotzinger, 2018-01-16:


Schneider1, 2020-05-13:
I did first tests with pyvisa-py under Ubuntu and it worked surprisingly well. It is not as fault tolerant as NI-Visa, and some devices do not strictly follow the protocol specifications, so sometimes a workaround will be needed. But in general, I think this is doable.

When touching all drivers to make them compatible with pyvisa-py, we should also optimize for pyvisa>1.5, i.e. also include #37, even if this means to give up compatibility with pyvisa<1.5

Schneider1, 2020-12-10:
In general, pyvisa-py works and I just merged #83 to add some support. However, visa.ask is no longer supported, so #37 is still open since many drivers are not changed.

closed 2020-12-10 by Schneider1