qkitgroup/qkit #33: measure_3D() fails after some (random) time: 'RuntimeError: Unable to register datatype id (Can't insert duplicate key)'

LGruenhaupt, 2018-01-29:
Unspecified run-time error.
Unable to register datatype id (Can't insert duplicate key)
d:20180124\202410_3D_chip3_antenna1_eskalation\202410_3D_chip3_antenna1_eskalation.h5

Exception in thread Thread-5:
Traceback (most recent call last):
  File "C:\Python27\Lib\threading.py", line 810, in __bootstrap_inner
    self.run()
  File "C:\Python27\Lib\threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\qtlab\qkit\qkit\gui\plot\plot.py", line 56, in save_plots
    h5plot(h5_filepath, comment=comment, save_pdf=save_pdf)
  File "C:\qtlab\qkit\qkit\gui\plot\plot.py", line 78, in __init__
    self.hf = hdf_lib.Data(path=self.path)
  File "C:\qtlab\qkit\qkit\storage\hdf_lib.py", line 46, in __init__
    self.hf = H5_file(self._filepath)
  File "C:\qtlab\qkit\qkit\storage\hdf_file.py", line 27, in __init__
    if self.hf.attrs.get("qt-file",None) or self.hf.attrs.get("qkit",None):
  File "C:\Python27\lib\site-packages\h5py\_hl\base.py", line 310, in get
    return self[name]
  File "C:\Python27\lib\site-packages\h5py\_hl\attrs.py", line 53, in __getitem__
    tid = attr.get_type()
  File "h5a.pyx", line 421, in h5py.h5a.AttrID.get_type (h5py\h5a.c:5340)
RuntimeError: Unable to register datatype id (Can't insert duplicate key)

Schneider1, 2018-04-27:
Is this still an issue?

Schneider1, 2020-05-27:
Since nobody complained about this any more, I'll close this issue.

Please feel free to re-open it, once you experience a similar behavior

closed 2020-05-27 by Schneider1