qkitgroup/qkit #39: qt.commands?

TimWolz, 2018-04-19:
Are the qt.commands, e.g., qt.mstart, qt.mend, ..., still needed?

Schneider1, 2018-04-27:
The former qtflow commands will get new functionality in qkit.flow.[start,end,sleep]. Please keep the old qt.mstart etc. to be able to replace them.

Apart from that, please replace `qt.instruments.create` with `qkit.instruments.create` if you work in a file, and remove the `import qt` if no longer needed.

yschoen, 2018-05-18:
Better to do it more explicitly by importing the used class directly. qkit.instruments is only a shortcut defined in S30_qkit_start.

To reduce confusion between instrument and instruments, the two modules have been renamed to instrument_base and instrument_tools by HR.

The old files exist still for legacy reasons in the master branch so far, but are already removed in the qtlab_core branch.

from qkit.core.instrument_tools import Insttools
Insttools.create()

Schneider1, 2018-05-18:
I agree that it should be safe to delete the instrument and instruments file, as we have in `S30_qkit_start.py`

``` python
from qkit.core.instrument_base import Instrument 
from qkit.core.instrument_tools import Insttools

qkit.instrument  = Instrument
qkit.instruments = Insttools()

if qkit.cfg.get("qt_compatible",True):
     [...]
    qt.instrument  = qkit.instrument
    qt.instruments = qkit.instruments
```
So even if we have old drivers, we do not need the instrument(s) files because we have instrument_base/tools.

However we **have to take  `qkit.instruments`** instead of instanciating `Insttools`: We create **one single instance** of Insttools and this instance tracks all instruments that we create. If you would create a new instance every time, there would not be a single list of instruments (i.e. `qkit.instruments.get_instruments()` would fail), which we need to store all settings of all devices in the `.set` file.

So please continue using `qkit.instruments.create`!

rotzinger, 2018-05-18:
> So please continue using qkit.instruments.create!
For the time being, agreed, Andre. 
However, I'm planning to replace it with a refactored qkit.dev.xyz mechanism, which also includes the device-db.    Have a nice weekend, Hannes

yschoen, 2018-05-23:
In the qtlab_core branch the qt commands are converted to:
qkit.flow.start
qkit.flow.sleep
qkit.flow.end

These are shortcuts to the functions in qkit.flow an instance of the qkit.core.flow.FlowControl class instanciated at qkit start.

yschoen, 2018-05-23:
Checkout qtlab_core branch to test the qkit version without qt references.

yschoen, 2018-09-04:
The branch has been merged to master and I consider this issue closed. Qt references are deprecated now and need to be explicitly turned on in the config. This should however be unnecessary as the equivalent qkit commands should work. The flow routines are in the future handled by core/flow. So far, they are mostly placeholders.

closed 2018-09-04 by yschoen