qkitgroup/qkit #40: Removed progress bar does not vanish completely

fr34q, 2018-04-27:
If a progress bar is replaced by a new one (e.g. in nested loops) the old one will still need some space in the notebook (tested in JupyterLab) so the new bar is placed a bit beneath creating more and more empty space in nested loops.

Example:
[image]

Expected behavior: Progress bar vanishes (display: none) or is even removed from DOM entirely.

Schneider1, 2018-04-30:
This is a bug of jupyterwidgets. Closed widgets leave an empty `<div>` container which is displayed by a blank space.

Solutions:
- fix the bug in jupyterwidgets itself
- do some JS magic to close the unneeded `<div>` cells (as I did it for the ipython notebook way of progressbar)
- Keep better track of all openend progressbars and always call `clear_output()`and display all progressbars
- **Quick fix**: Almost all routines that (I have written and) open a progress bar have the option `progress_bar = False` to supress the creation for PBs in the inner loop. I think for 1e5 iterations in 11secs, as shown in the picture, a PB is not needed

fr34q, 2018-05-02:
Issue described here: https://github.com/jupyter-widgets/ipywidgets/issues/1845
and fixed in release 7.2 of ipywidgets

Therefore, to fix the issue upgrading local ipywidgets installation is sufficient:
`pip install --upgrade ipywidgets`

closed 2018-05-02 by fr34q