qkitgroup/qkit #44: Qviewkit automatically closes for 2D time domain data

TimWolz, 2018-05-02:
I can't display my 2d time domain data with qviewkit. When I click on amplitude or phase in the gui. The plot window shortly opens but then the complete qui closes...

Schneider1, 2018-05-02:
Please provide a traceback from command line or a sample file.

TimWolz, 2018-05-02:
There you go
[P83GHM_time.zip](https://github.com/qkitgroup/qkit/files/1968156/P83GHM_time.zip)

TimWolz, 2018-05-02:
and Traceback:

>  from ._conv import register_converters as _register_converters
> Traceback (most recent call last):
>   File "C:\ProgramData\Anaconda2\lib\site-packages\pyqtgraph\imageview\ImageView.py", line 567, in roiChanged
>     data, coords = self.roi.getArrayRegion(image.view(np.ndarray), self.imageItem, axes, returnMappedCoords=True)
>   File "C:\ProgramData\Anaconda2\lib\site-packages\pyqtgraph\graphicsItems\ROI.py", line 1110, in getArrayRegion
>     result, coords = fn.affineSlice(data, shape=shape, vectors=vectors, origin=origin, axes=axes, **kwds)
>   File "C:\ProgramData\Anaconda2\lib\site-packages\pyqtgraph\functions.py", line 491, in affineSlice
>     grid = np.mgrid[tuple([slice(0,x) for x in shape])]  ## mesh grid of indexes
>   File "C:\ProgramData\Anaconda2\lib\site-packages\numpy\lib\index_tricks.py", line 187, in __getitem__
>     nn = _nx.indices(size, typ)
>   File "C:\ProgramData\Anaconda2\lib\site-packages\numpy\core\numeric.py", line 1853, in indices
>     res = empty((N,)+dimensions, dtype=dtype)
> MemoryError

mpfirrmann, 2018-05-03:
I can confirm the issue on my office computer in the master branch. The file opens in the qvk_2 branch where we work on qviewkit but I don't know (yet?) why this is no issue there. I've had some troubles with pyqtgraph's ROI in the past, IDK if we need this in qviewkit at all.

commit 5a5fc35e (2018-05-03): Fixes #44: qviewkit roi bugfix (pyqtgraph issue)
closed 2018-05-03 by rotzinger