qkitgroup/qkit #46: Write HOWTO - jupyter notebooks on simple things

rotzinger, 2018-05-05:
Hi,
just that we don't forget --  we should add more HOWTO jupyter notebooks to 
qkit/doc/notebooks with "simple" things like
howto use a sample object
howto use qfit 
howto use timedomain, transport, spectroscopy ...
(...)

They should be simple and short and with some explanation what is done.

Cheers, Hannes

mpfirrmann, 2018-05-11:
-h5_readout (init, file_database, data and attributes as object parameters, open viewer)
-basic measurements: spectroscopy, transport, timedomain
-create/update/deleter instruments ("new" qkit-call)
-qkit config
-sample object
-measurement object (re-create the measurement environment of a previous measurement)
-usage of (all?) analysis routines that are in qkit

-show some features of the viewer? maybe not with notebooks but that would still be useful

mpfirrmann, 2018-05-11:
-qplexkit?
-qsurveilkit?