qkitgroup/qkit #50: add possibility to select slices in qkview views

Schneider1, 2018-05-15:
Example:
If one makes a 2D VNA measurement with only one or three points in the VNA and second parameter, one wants to view the amplitude/phase of the center point of the VNA against the second parameter. The amplitude/phase however are 2D datasets, so they will be plotted as colorplots.
Currently, we save the amplitude/phase of the center point in a second dataset, which is displayed then.

It would be more elegant to have the possibility to select a slice of the data and display it as a view. We already have the `filter` attribute in hdf views, which could maybe be used for this purpose, but as far as I've seen, there are no filters implemented.

mpfirrmann, 2018-05-15:
Another possible way to solve this would be similar to the options provided for the value_boxes. Here it is possible to slice the 3d matrix from all directions and manually change the parameters for the slice. I could imagine something similar to that. Of course, depending on the slicing direction this would produce only more or less a comprehensible display of the data.
Nevertheless, I feel that in some way this would be more consistent compared to adding a filter attribute to the views.
Any thoughts or opinions on that?

Schneider1, 2018-05-16:
I totally agree that it would be great to select the slicing direction in the plot window. If we would manage to open this view by default on the right slicing direction and the central point, this would fully solve the issue. But I guess we would need an attribute in our h5 file to tell the viewer that he sould do so?

mpfirrmann, 2018-05-16:
Added new feature with commits 4fd6895, 8319ae6, 13832c5, 5bb4822 in the qvk_2 branch. Please test these added features (and the whole changes in qvk_2) so that we can merge this branch to master in the near future.

closed 2018-05-16 by mpfirrmann