qkitgroup/qkit #55: Traceback for spectroscopy measure 2D and 3D

TimWolz, 2018-06-12:
During 2D and 3D measurments in spectroscopy, probably also in measure_td and transport?, no full traceback is provided, which makes debugging absolutely tedious.
[image]

Schneider1, 2018-06-12:
This comes from the very crude `except Exception as e` in spectroscopy.py:644 https://github.com/qkitgroup/qkit/blob/da5c13d33cc488c24173cd8f8752ad10c0c29e6f/qkit/measure/spectroscopy/spectroscopy.py#L644

I suggest that the user should catch any error if he wants to run measurements in a loop and we simply remove the `except` block while maintaining the `finally` block.

rotzinger, 2018-07-04:
Is this appropriately answered ?

commit ed951534 (2018-07-05): spectroscopy and transport: removed exception handling. Closes #55
closed 2018-09-04 by Schneider1