qkitgroup/qkit #57: data saving interface for non-standard experiments

TimWolz, 2018-07-04:
Would it make sense to have an interface for data saving if we perform non-standard experiments? 

By that I mean: so far data are saved via spectroscopy, measure_td, transport. All those scripts are specifically coded for some standard experiments, e.g., resonator readout. Amplitude/Phase traces. However sometimes one wants to measure different things and often it is only a few commands to get the data but to save them in h5-files with all the functionality qkit provides takes many more commands.
What I would appreciate is to have a function/class where I can simply pass my previously extracted data as well as the coordinate names. The rest, e.g, time stamp, instrument_settings,..., should then be handled automatically.

rotzinger, 2018-07-04:
Tim, do I get it right, the  methods described  in  qkit/doc/notebooks/store_data_basics.ipynb
or a subset is still to complicated? Well I guess one could write such a class, but it most likely would be limited in its power.
Cheers, Hannes

TimWolz, 2018-07-05:
Well it's not too complicated but it's some commands and at least I have to look them up. I was more or less thinking of an easy quick save. But I think you're right. Such a class would be too limited in its power. I'll close the issue.

closed 2018-07-05 by TimWolz