qkitgroup/qkit #69: Automatic index axis for hdf_dataset?

fr34q, 2019-02-22:
Is it possible to have an implicit index axis when using hdf.add_value_vector() ?

If I store 1mio data points in the hdf file and then also have to store the same amount of "index" values in a separate axis, my file size doubles...

So, with data axis: Nice x axis for the data points, but wasted disk space
Without data axis: Plot displays but Axis is labeled _none_ (M_none_)

Creating the axis and not specifying any values for it also doesn't work, same result as without axis.

In both cases I also get:
```
Exception in qkit/gui/plot/plot.py while plotting
/entry/data0/i_component
'h5plot' object has no attribute 'x_exp'
```

fr34q, 2019-02-22:
Relevant code snippet:
```
self._data_file = hdf.Data(name=self.dirname, mode='a')
self._data_axis = self._data_file.add_coordinate(
    self.x_coordname, unit=self.x_unit)

# How can I avoid this?
self._data_axis.add(np.arange(count, dtype=int))

self._data_i = self._data_file.add_value_vector(
            'i_component', x=self._data_axis, unit=self.data_unit)

self._data_i.append(data_i)

self._data_file.close_file()
```

rotzinger, 2019-02-22:
Hi,
yes, in the moment there is no way around having a second dataset for the x-label.

We had this "feature" in the past, where the labels have been stored with the dataset.  Now the label for the axis is taken via the "x|y url"-attribute and thus from the associated dataset. 

As an easy fix, we could change the default label for an "index" plot to something more sensible
like "index" in Qviewkit. 
Would this serve you?

fr34q, 2019-03-19:
@rotzinger Thanks for the reply and sorry for my delay. 

So my expectation would be that the axis is the same as for my other axes values e.g. "current"/"duration" etc. where the values are plotted over "index (#)".

I am not sure if changing the default label will work out to be the same as there is still an undefined property (x=None) of the data there that should be handled appropriately - what are your thoughts?