qkitgroup/qkit #77: FID: Error when pandas is not installed

fr34q, 2019-12-16:
When starting qkit, I get this error:
```
Exception in thread creating_db:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/rgebauer/phd/projects/qkit/qkit/core/lib/file_service/file_info_database.py", line 190, in update_all
    self.update_grid_db()
  File "/home/rgebauer/phd/projects/qkit/qkit/core/lib/file_service/file_info_database.py", line 196, in update_grid_db
    self._initiate_basic_df()
  File "/home/rgebauer/phd/projects/qkit/qkit/core/lib/file_service/file_info_database.py", line 209, in _initiate_basic_df
    self.df = pd.DataFrame(self.h5_info_db).T
NameError: name 'pd' is not defined
```
Turns out I hadn't had panda installed. Having a quick look at the File info database (FID) script, it looks like pandas is optional (in a try catch block), so I guess this error should not appear.

rotzinger, 2019-12-16:
Hey Richard,

yes that's a bug and has to be fixed.
In the meantime, can you just install pandas?

fr34q, 2019-12-16:
Sure, that's what I did. I just wanted to write down this edge case somewhere as it is not experienced that often.

closed 2020-05-27 by Schneider1