qkitgroup/qkit #78: Big viewer bug when zooming

TimWolz, 2020-04-03:
The x-Axis does not scale correctly in a line plot when zooming in or out. See attached screenshots. Basically the resonance frequency dip is at two different frequencies. Depending on the zoom level it can be tens of MHz apart. 
Especially if you don't know this bug and want to have a quick look for frequency changes it can be really confusing.
Issue might be python3 related.
[image]
[image]

rotzinger, 2020-04-07:
Could not be verified.

closed 2020-04-07 by rotzinger

Schneider1, 2020-04-07:
It's a tricky one: It only appears when you have two screens with different scaling setting in Windows. This is mostly relevant when you work on a laptop with external screen connected.

It is pretty obvious that different elements in the graph get scaled differently, i.e. the axis and the displayed data do not have the same size and therefore do not match.

The bug is already known in pyqtgraph https://github.com/pyqtgraph/pyqtgraph/issues/756

commit 90736270 (2020-12-01): Qviewkit: Fix #78 bug when operating on Windows with multiple screens of different scaling