qkitgroup/qkit #79: Progressbar and other widgets not visible in jupyter lab

Schneider1, 2020-04-22:
We noticed that on `jupyter lab` the progress bar and other ipywidgets like the query for username and runID at startup are not visible and replaced by a text (see image below)
[image]
If we start `jupyter notebook` instead and execute the same notebook, it works fine.

Schneider1, 2020-04-22:
According to [this issue](https://stackoverflow.com/questions/49542417/how-to-get-ipywidgets-working-in-jupyter-lab) jupyter**lab** has a more strict policy towards foreign JavaScript code than jupyter**notebook**, and you need to register the widgets as extension. For this, you need `nodejs` installed and `jupyter lab` has to be set up properly.

Running the following in a shell shold work (make sure to activate your productive conda environment beforehand) 
```
conda install nodejs
conda install jupyterlab
jupyter labextension install @jupyter-widgets/jupyterlab-manager
```

closed 2020-04-22 by Schneider1

JBraumueller, 2020-04-22:
I had the same issue with our ProgressBar and jupyter lab. As you describe it is resolved by installing nodejs, the ipywidgets, and maybe updating jupyter lab.