qkitgroup/qkit #91: KeyError: 'shutdown_hook' when starting Qkit in new Jupyter Notebook

fr34q, 2022-02-25:
I updated JupyterLab to version 3.2.8 and now starting Qkit (`qkit.start()`) fails with the following error:
```
File ~/phd/projects/qkit/qkit/core/s_init/S30_qkit_start.py:36, in <module>
     33 # Other functions should be registered using qt.flow.register_exit_handler
     34 from qkit.core.lib.misc import register_exit
---> 36 register_exit(flow.qtlab_exit)

File ~/phd/projects/qkit/qkit/core/lib/misc.py:49, in register_exit(func)
     47 ip = get_ipython()
     48 if ipython_is_newer((0, 11)):
---> 49     ip.hooks['shutdown_hook'].add(func, 1)
     50 else:
     51     ip.IP.hooks.shutdown_hook.add(func, 1)

KeyError: 'shutdown_hook'
```

Maybe the key was removed or renamed?

treisinger42, 2022-03-11:
ipython version 8 appears to be the problem. the key 'shutdown_hook' was removed:
https://ipython.readthedocs.io/en/stable/whatsnew/version8.html

Downgrading to ver 7 works so far:
pip install ipython==7.31 --force-reinstall

closed 2022-03-15 by Schneider1

Schneider1, 2022-03-15:
Yes, in the newer IPython versions, they removed support for some older structures. I did not really find where the shutdown hooks moved to, it seems that this functionality was just removed.

However, we can use the `atexit` module, which also calls a functions when exiting the kernel (I just tested it).

However2, what we are doing in the atexit call is simply to `print("Closing QKIT...")`. I guess we could have lived without this functionality. Still, it's good to have the possibility to do something more meaningful here.

commit 2524c18d (2022-03-15): Add atexit handler for ipython > 8.0. resolves #91