qkitgroup/qkit #108: Recent change to resonator analysis class breaks circle fit in 2D Measurements.

thilokru, 2023-05-31:
See comment https://github.com/qkitgroup/qkit/commit/c93b1c6d94d7858e67a519b58e455d7daea146c9#r115820757 .
Setting `i=1` at the beginning of the loop fixes the problem, but this an ugly hack.

@Schneider1

paoloaq, 2024-11-23:
Hi @thilokru,

I am trying to use Qkit to perform a fit. However, after applying the changes from commit #c93b1c6, I am encountering the following exception when calling the fit_circle method of Resonator: 

```` python
Resonator' object has no attribute '_circ_amp_gen'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[26], line 6
      2 res = Resonator(h5d.get_filepath())
      4 #res.fit_lorentzian(fit_all=True,f_min=5.667e9,f_max=5.668e9)
      5 #res.fit_fano(fit_all=True)
----> 6 res.fit_circle(fit_all=True,f_max=5.668e9)
      9 '''
     10 res.set_x_coord(s33_data[:, 0])
     11 frq_low=7.740
   (...)
     15 c.plotall()
     16 '''        

File ~/git/qkit_mod/src/qkit/analysis/resonator.py:232, in Resonator.fit_circle(self, reflection, notch, fit_all, f_min, f_max)
    229 self._prepare_f_range(f_min, f_max)
    231 if self._first_circle:
--> 232     self._prepare_circle()
    233     self._first_circle = False
    235 if self._circle_reflection:

File ~/git/qkit_mod/src/qkit/analysis/resonator.py:345, in Resonator._prepare_circle(self)
    342         self._results[key] = self._hf.add_value_vector('circ_' + key, folder='analysis', x=self._x_co, unit='')
    344 circ_view_amp = self._hf.add_view('circ_amp', x=self._y_co, y=self._ds_amp)
--> 345 circ_view_amp.add(x=self._frequency_co, y=self._circ_amp_gen)
    346 circ_view_pha = self._hf.add_view('circ_pha', x=self._y_co, y=self._ds_pha)
    347 circ_view_pha.add(x=self._frequency_co, y=self._circ_pha_gen)

AttributeError: 'Resonator' object has no attribute '_circ_amp_gen'"
````

Thank you for any help. 
Best regards

thilokru, 2024-11-25:
@paoloaq What qkit.cfg['circle_fit_version'] are you using?

paoloaq, 2024-11-25:
I receive the same exception regardless of the circle_fit_version setting (1 or 2)

paoloaq, 2024-11-27:
Hi @thilokru, could you tell me from which commit the issue with the initialization of the Resonator attributes might have been introduced? 
I would like to try the most recent version possible that is not affected by the issue.

Thank you so much